MMRIZE/MMM-GroveGestures #4: multiple notification

bcoleg, 2019-02-22:
First off, Thanks so much eouia!  I've been using several of your modules on my mirror.  

what is the proper way to switch the set of commands?  This is what I am trying to do, but it isn't working.

`        "FORWARD-BACKWARD": {
          notificationExec: {
            notification: "NEWS_DETAIL",
            payload: null,
          },
            notificationExec: {
            notification: "GG_CHANGE_COMMANDSET_NEWSMODE",
            payload: null,
          },
        },`

eouia, 2019-02-22:
You have to understand these;

- First,`commandSet` is some mode which might have diffent commands,
- Module can select one SET of `commandSet`s to be used by `commandSetTrigger`,
- `commandSetTrigger` is definition of `INCOMING NOTIFICATION`. When specific `notification` is received, module change its commandSet by condition of that notification.

By example;
MMM-Pages(https://github.com/edward-shen/MMM-pages) module emit `PAGE_CHANGED` notification when it's page is changed.
If you need commandSet `COMMANDSET_A` for page 1, and need another commandSet `COMMANDSET_B` for page 2, You can make `commandSetTrigger` like this.
```js
commandSetTrigger: {
  "PAGE_CHANGED" : (payload) => {
    if (payload == 1) return "COMMANDSET_A"
    if (payload == 2) return "COMMANDSET_B"
  }
}
```
This code means, If "PAGE_CHANGED" notification is comming, check its payload. if the payload equal 1, COMMANDSET_A is commandSet name to use. if the payload equal 2, COMMANDSET_B is that.

eouia, 2019-03-01:
Have you still got a problem? I'll close this issue once because there is no more feedback. You can ask me again anytime.

closed 2019-03-01 by eouia