MMRIZE/MMM-GroveGestures #5: Keyboard Interrupted ???

ezar, 2019-03-03:
Hi eouia, 

Everytime appears this error, and I cannt use it :(
Why?

1|MagicMir | Keyboard Interrupted
1|MagicMir | [GESTURE] Gesture script is finished.

ezar, 2019-03-03:
Can you restart pyshell when onError?

eouia, 2019-03-04:
When this interrupt is appeared? This error would be happened when you try to kill the script by manual. Describe me more details what you tried, and what you expected as a result of your trial, but how it failed.

ezar, 2019-03-04:
Appears after start MagicMirror. I didn't nothing with the script.
When it appear, I need to restart MagicMirror because Gesture doesn't works :(

eouia, 2019-03-04:
It might mean your python script has an error from start with your seeting. Check it works or not

eouia, 2019-03-04:
https://github.com/eouia/MMM-GroveGestures/blob/master/README.md#installation-of-module

Did you all the steps by instructions above link?
There be also an instruction how to check your script

ezar, 2019-03-04:
Yes! The script works sometimes. Other times gets timeout, others error :(

ezar, 2019-03-04:
Paj7620 initialize register finished.
Traceback (most recent call last):
  File "gesture_print.py", line 43, in <module>
    g.print_gesture()
  File "/home/pi/MagicMirror/modules/MMM-GroveGestures/py/grove_gesture_sensor.py", line 440, in print_gesture
    data1=self.paj7620ReadReg(0x44, 1)[0]
  File "/home/pi/MagicMirror/modules/MMM-GroveGestures/py/grove_gesture_sensor.py", line 372, in paj7620ReadReg
    return bus.read_i2c_block_data(self.PAJ7620_ID, addr,qty)
IOError: [Errno 121] Remote I/O error

bolish, 2019-03-04:
Hi,
I also get error. (sometimes MMM-GroveGestures stops working or even don't start working).
How do you get the error log for analysis purpose? I would like to share my issue also.

ezar, 2019-03-04:
I activate using config.

ezar, 2019-03-04:
Seems that we have a more recent version

https://github.com/Seeed-Studio/grove.py/blob/master/grove/grove_gesture_sensor.py

eouia, 2019-03-05:
Of course, I tested it with RPI(RPI 3 B+)I also. Hmmmm....
I'll try recent version script in a few days. (I'm on vacation now, sorry)

eouia, 2019-03-05:
Ok. I've tested my module again with newly formatted RPI from H/W installation.
Environment :
- RPI 3 B+
- Raspbian Stretch with desktop
  - Version:November 2018
  - Release date:2018-11-13
  - Kernel version:4.14
  - SHA-256:a121652937ccde1c2583fe77d1caec407f2cd248327df2901e4716649ac9bc97
- node : v10.15.1
- npm : v6.4.1
- Python : 2.7.13
- Python3 : 3.5.3 (Python 2 and Python 3 both work)

My module(and py scripts) is working as expectation. 

I doubt you've missed or failed something on H/W installation. Could you check again?
Especially, `sudo i2cdetect -y 1` should show you `73` on the matrix of result.

eouia, 2019-03-05:
If sometime work or not, It seems your GPIO(I2C) connection is not so stable. Maybe it could be H/W problem or Electricity issue.

ezar, 2019-03-08:
- No problem H/W. 
- No Electricity issue.

I'm using a I2C temperature sensor without problem :(

eouia, 2019-03-10:
Sorry, but I cannot find any suspicious point at this moment.
Does `'i2cdetect -y 1' ` show proper address of sensor?(`73`)?
Usually `IOError: [Errno 121] Remote I/O error` itself occurs when a not available address is referenced.  And usually it come from not installed or connected H/W. (Of course there could be other reasons... however,)

ezar, 2019-03-10:
Seem solved ;)
I reduce i2c speed to 10000.
Regards

eouia, 2019-03-10:
Nice to hear it. I haven’t think about it. I’ll close this issue.

closed 2019-03-10 by eouia

jonathanbonette, 2024-04-27:
> Seem solved ;) I reduce i2c speed to 10000. Regards

works for me as well, for anyone interest:

in terminal:
**sudo nano /boot/firmware/config.txt**
if you dont have this line in your config.txt, add it right below to the others dtparam configs
**dtparam=i2c_arm_baudrate=10000**
ctrl + x to exit, y to confirm, and enter to save it
reboot your rasp
**sudo reboot**
after the reboot, just check the file to see if it is saved correctly