MMRIZE/MMM-GroveGestures #6: possible to use it like a PIR?

smiley-2, 2019-03-12:
Hello,In first, sorry for my english.
I think to buy this sensor to use ith your module.

It's possible to use your module like a PIR sensor?
Per example, when the grove detect a gesture, the screen power on and if no gesture for a time, the screen power off?

thanks for your help.

eouia, 2019-03-13:
Of course possible. This module has “onIdle” features, and ability for execution shell commands or module method or notification. The combination of features will work

smiley-2, 2019-03-13:
thank you, you have some example? please?

eouia, 2019-03-13:
```js
onIdle: {
  shellExec: "YOUR MONITOR POWER OFF SHELL COMMAND" // this command will be different per your environment 
},
onDetected: {
  shellExec: "YOUR MONITOR POWER ON SHELL COMMAND"
},
```
Add that in configuration of this module.

eouia, 2019-03-13:
Or you can assign specific gesture pattern to turn on monitor.

smiley-2, 2019-03-14:
great !

thank you!

eouia, 2019-03-15:
I'll close this. have a nice day.

closed 2019-03-15 by eouia