MMRIZE/MMM-GroveGestures #7: Compatibility with other sensors (Seeed Flick)

bolish, 2019-03-13:
Hi @eouia 

Did you tested your module with other sensors as well?

i.e : i would like to use the [Seeed Flick](https://www.seeedstudio.com/Flick-Zero-3D-Tracking-Gesture-pHAT-for-Raspberry-Pi-Zero-p-2950.html) senor, which seems pretty close (for a non expert like me) to the functionning of the Grovegesture one.

Do you think there is any chance to make it work?

(I prefer to check before buying it...)

Thanks in advance

eouia, 2019-03-13:
I think not working without modification. And I have no idea about how the modification will be hard at this time.
If I had one, I can test. And I believe I can make one module for that sensor without difficulty. But I have not. :D

Different topic.
HAT version should be topped on RPI. I don't like HAT version because it makes to give bad effect for heat control.
And Many HATs are not compatible with non-RPI devices.
At last, To use it as MM controller, the physical location of this sensor could makes some headache on assembly of MM frame.
Those are why I didn't try HAT version of flick. (But I have interesting about FLICK LARGE. Maybe someday I'll try it.)

bolish, 2019-03-15:
Hi Sean,
Fully agree about the HAT thing. It was just the beginning of my search but as I'm also using GPIO for other sensor, the HAT would have been nogo.
Indeed, the Large version should be a better option.

Do you think that Flick is really better compared to PAJ7620u2? I mean, will it be worth investing time / money (will it really improve user experience?)

Buying it is not really the issue on my side, but I know I will be blocked by the coding part!!

eouia, 2019-03-15:
I don't know much about Flick.

Comarison... hmmmm...

- Flick has touch/tap feature. It could be good to give you more gesture. And touch/tap on board is solid than swaying on air.
- But it should be placed somewhere, on your mirror frame, or else other location. You should consider position or assembly on your mirror.
- Sensor's specification is almost similar. I don't know which is beter to detect gesture.(I didn't test)

eouia, 2019-03-23:
Have you any further question? I'll close this.

closed 2019-03-23 by eouia