MMRIZE/MMM-GroveGestures #8: running error:

Pengree, 2019-03-16:
@eouia 

Very thanks for you to give us the 3D tracking & gesture solution on MagicMirror!
According to your steps, I have done all configuration. Now I passed the gesture test:

```
python gesture_print.py
wake-up finish.
Paj7620 initialize register finished.
Down
Up
Down
Down
Left
Up
Right
```
And I added some config in config.js. When I start MagicMirror by command npm start, the screen appeared the following error log:
```
2019-03-16T09:39:58 <log> Whoops! There was an uncaught exception... (/home/pi/MagicMirror/js/app.js:34 process.<anonymous>)
2019-03-16T09:39:58 <error> { Error: spawn EACCES
    at _errnoException (util.js:1024:11)
    at ChildProcess.spawn (internal/child_process.js:323:11)
    at Object.exports.spawn (child_process.js:514:9)
    at new PythonShell (/home/pi/MagicMirror/modules/MMM-GroveGestures/node_modules/python-shell/index.js:96:45)
    at Class.job (/home/pi/MagicMirror/modules/MMM-GroveGestures/node_helper.js:52:18)
    at Class.socketNotificationReceived (/home/pi/MagicMirror/modules/MMM-GroveGestures/node_helper.js:26:14)
    at Socket.<anonymous> (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:113:11)
    at emitTwo (events.js:126:13)
    at Socket.emit (events.js:214:7)
    at /home/pi/MagicMirror/node_modules/socket.io/lib/socket.js:528:12 code: 'EACCES', errno: 'EACCES', syscall: 'spawn' } (/home/pi/MagicMirror/js/app.js:35 process.<anonymous>)
```

I guess that python-shell or other things causes this issue. Could you help me? I'd appreciate it if you can give me some suggestions.

Have a good weekend!

eouia, 2019-03-16:
That error occurs when script permission is wrong. Maybe your Mirror is running as different permission with python script(I don’t know why or what exactly)
Give proper permission to scripts.

Pengree, 2019-03-17:
> That error occurs when script permission is wrong. Maybe your Mirror is running as different permission with python script(I don’t know why or what exactly)
> Give proper permission to scripts.

@eouia 
Thanks for your reply!
Yes, I modified some scripts permission and the issue disappeared! 

Let me go ahead to enjoy your fantastic module.
Thanks again!

closed 2019-03-17 by Pengree