MMRIZE/MMM-GroveGestures #9: MMM-GroveGestures module automatically stop running

Pengree, 2019-03-19:
@eouia 

I found that the  module automatically stop running. The console log is as followings:
```
2019-03-18T22:54:47 <log> Keyboard Interrupted (/home/pi/MagicMirror/modules/MMM-GroveGestures/node_helper.js:39 Class.log)
2019-03-18T22:54:47 <log> [GESTURE] Gesture script is finished. (/home/pi/MagicMirror/modules/MMM-GroveGestures/node_helper.js:39 Class.log)
```
There are not keyboard or mouse in Raspberry Pi 3b+. During the whole process, I did not touch any windows from remote ssh session.

The config is very simple. There is one module each page. the config is as following:
```
	modules: [
		{
            module: 'MMM-Logging',
            config: {
            }
        },

		{
			module: "alert",
			config: {
			}
		},

		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: 'calendar_monthly',
			position: 'top_left',
			config: {
			}
		},
		{
			module: "calendar",
			header: "China Holiday",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://p64-calendars.icloud.com/holidays/cn_zh.ics"
					}
				],
				maximumEntries: 5
			}
		},

		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "XXXX",
				locationID: "179XXX",  //ID from http://bulk.openweathermap.org/sample/; unzip the gz file and find your city
				appid: "9fa3e89aXXXXXXXXX1a614db351"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast：",
			config: {
				location: "1796XXX",
				locationID: "179XXX",  //ID from https://openweathermap.org/city
				appid: "9fa3e89a8eXXXXXbb261a614db351",
				maxNumberOfDays: 7
			}
		},

		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Zhihu Daily",
						url: "https://rsshub.app/zhihu/hotlist",
					},
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
		{
			module: 'MMM-pages',
			config: {
				modules:
					[
						["alert", "updatenotification", "clock", "newsfeed"],
						["currentweather","weatherforecast"],
						["calendar_monthly","calendar"],
						["MMM-Memo"],

						
					],
				excludes: 
					["MMM-pages", "MMM-page-indicator", "MMM-GroveGestures"],
			}
		},
		{
			module: 'MMM-page-indicator',
			position: 'bottom_bar',
			config: {
				pages: 5,
			}
		},

		{
			module: 'MMM-Memo',
			position: 'bottom_left',
			config: {
				memoTitle: "Homeworks",
				memoDisplayIfEmpty: true,
				memoDisplayHeader: true,
				memoDisplayId: true,
				memoColorHeader: 'Blue',
				memoRotation: '-8',
				memoWidth: '180px',
				memoHeight: '380px',
				memoMaxItems: 26
			}
		},

		{
			module: "MMM-GroveGestures",
			position: "middle_center",
			config: {
				autoStart: true, //When Mirror starts, recognition will start.
				verbose: true, // If set as `true`, useful messages will be logged.
				recognitionTimeout: 1000, //Gesture sequence will be ended after this time from last recognized gesture.
				cancelGesture: "WAVE", //If set, You can cancel gesture sequence with this gesture.
				visible: true, //Recognized gesture sequence will be displayed on position	  
				idleTimer: 1000*60*30, // `0` for disable, After this time from last gesture, onIdle will be executed.
				onIdle: { // See command section
					moduleExec: {
						module: [],
						exec: (module, gestures) => {
							module.hide(1000, null, {lockstring:"GESTURE"});
						}
					}
				},
				onDetected: {
					notificationExec: {
						notification: "GESTURE_DETECTED",
					},
					// You can make Mirror to wake up the modules which were hidden by onIdle with any gestures.
					moduleExec: {
						module: [],
						exec: (module) => {
							module.show(1000, null, {lockstring:"GESTURE"});
						}
					}
				},

				gestureMapFromTo: { //When your sensor is installed with rotated direction, you can calibrate with this.
					"Up": "UP",
					"Down": "DOWN",
					"Left": "LEFT",
					"Right": "RIGHT",
					"Forward": "FORWARD",
					"Backward": "BACKWARD",
					"Clockwise": "CLOCKWISE",
					"anti-clockwise": "ANTICLOCKWISE",
					"wave": "WAVE"
				},

				defaultNotification: "GESTURE",
				pythonPath: "/usr/bin/python3", // your python path

				defaultCommandSet: "default",
				commandSet: {
					"default": {
						"FORWARD-BACKWARD": {
							notificationExec: {
								notification: "ASSISTANT_ACTIVATE",
								payload: null
							}
						},
						"LEFT-RIGHT": {
							notificationExec: {
								notification: "ASSISTANT_CLEAR",
								payload:null,
							}
						},
						"ANTICLOCKWISE": {
							moduleExec: {
								module: [],
								exec: (module, gestures) => {
									module.hide(1000, null, {lockstring:"GESTURE"});
								}
							}
						},
						"CLOCKWISE": {
							moduleExec: {
								module: [],
								exec: (module, gestures) => {
									module.show(1000, null, {lockstring:"GESTURE"});
								}
							}
						},
                        "LEFT": {
                            notificationExec: {
                                notification: "PAGE_INCREMENT",
                                payload: null,
                            }
                        },
                        "RIGHT": {
                            notificationExec: {
                                notification: "PAGE_DECREMENT",
                                payload: null,
                            }
                        },
                        "UP": {
                            notificationExec: {
                                notification: "ARTICLE_PREVIOUS",
                                payload: null,
                            }
                        },
                        "DOWN": {
                            notificationExec: {
                                notification: "ARTICLE_NEXT",
                                payload: null,
                            }
                        },

					},
				},
				commandSetTrigger: {
					"GG_CHANGE_COMMANDSET_NEWSMODE": "NEWS_MODE",
					"GG_CHANGE_COMMANDSET_BY_PAYLOAD": (payload) => { // You can use callback function to change set with conditional payload values.
						return payload.commandSetName;
					}
				},
			}
		},
        ]
```

How to troubleshooting this issue? Could you give me some suggestion? Thanks in advance!

Have a nice day!

eouia, 2019-03-19:
When did that error occur? on start? or some period after running?

Keyborad Interrupt means, Grove Gesture detection script (grove_gesture_sensor.py) was quitted by external interruption (Sorry for confusing, someday I'll edit the error message)
This error will occur when;
- User stop program by force (by example; Ctrl + C with keyboard)
- Your Sensor and RPI has a problem to connect through GPIO pin (bad assembly, electricity shortage, ...) => Check your assembly.

Well, I'll try to modify source to restart when that error occurs, Plz wait a few days.

Pengree, 2019-03-19:
@eouia 
Thanks for your suggestion. 

I'll check these connections although I don't think so. The PAJ7620u2 sensor was tightly connected Respeaker-2mic I2C interface by one 4-Pin cable. 
[image]

 This issue often happened during some period after MagicMirror running. NO KEYBOARD, NO TOUCH, even if NO SSH SESSION (disconnect ssh session).

According to the issue https://github.com/eouia/MMM-GroveGestures/issues/5, I modified i2c speed to 10000. Although after that the issue happened one time again, I found that the issue seems like disappear. I'm not sure. I'll pay attention on it.

closed 2019-03-19 by Pengree

eouia, 2019-03-19:
Maybe some i2c speed might make too fast or too slow connection between sensor and RPI. I didn't think about it before. I hope you'll get a good luck. (if so, I don't need to edit... :D)

Pengree, 2019-03-19:
If the issue will not happen again in future, I'll tell you.
Thanks for your fantastic application and your conscientious attitude!

xuepgao, 2019-03-20:
@eouia When the i2c speed is 10000, the issue unluckily appeared two times during 24 hours. Although it seems like that the frequency decreases, I still adjusted the i2c speed to 32000. The issue  hasn't happened in past 30 hours. Let me keep watching it.

Pengree, 2019-03-20:
@eouia When the i2c speed is 10000, the issue unluckily appeared two times during 24 hours. Although it seems like that the frequency decreases, I still adjusted the i2c speed to 32000. The issue hasn't happened in past 30 hours. Let me keep watching it.