MMRIZE/MMM-GroveGestures #10: Help

BenjaminH86, 2019-03-23:
Hello World,

I love your module!
I use MMM-page-selector to switch between 2 pages. I wanted to use grove gesture to change pages, but I can not adapt the script.
I can see my gestures in the Magic Mirror when I pass my hand in front of the grove gesture; after, i do not know what to put in config.js
Can you help me?
For information, my english is bad, and I do not know anything about programming!

Thank you

BenH

eouia, 2019-03-23:
Let's assume You have page `1` and `2`.

You can assign gesture `LEFT` to `1`, `RIGHT` to `2`.
At that case, you can add this to your command set(default or any other you want).
```
"LEFT": {
  notificationExec: {
    notification: "PAGE_SELECT",
    payload: "1",
  }
},
"RIGHT": {
  notificationExec: {
    notification: "PAGE_SELECT",
    payload: "2",
  }
},
```

The problem is, `MMM-page-selector` has no feature of "NEXT PAGE" or "PREVIOUS PAGE". so when there be more than 2 pages, you should assign gestures per each page.

BenjaminH86, 2019-03-23:
okay.
Thank you for your quick reply.
did i just say it in config, or should i set something else before? for the sensor to be active when Magic Mirror is started?

eouia, 2019-03-23:
In config. Read documents of this module carefully.

Pengree, 2019-04-12:
> Let's assume You have page `1` and `2`.
>
> You can assign gesture `LEFT` to `1`, `RIGHT` to `2`.
> At that case, you can add this to your command set(default or any other you want).
>
> ```
> "LEFT": { [...]

@eouia 
```
I'm sorry, you are wrong.
In my config.js, the following config works fine:
.....
    "LEFT": {
        notificationExec: {
            notification: "PAGE_UP",
            payload: null,
        }
        },
    "RIGHT": {
        notificationExec: {
        notification: "PAGE_DOWN",
        payload: null,
    }
......
```
`MMM-page-selector` has the feature of switching page notification.

eouia, 2019-04-24:
Have you got still a problem? I'll close this issue because there be no feed back any more. Wish a good luck.

closed 2019-04-24 by eouia