MMRIZE/MMM-GroveGestures #12: Questions about distance & swipe to another page

JohnB7777, 2019-06-03:
Hello,

I come from the forum of your "MMM-Hotwords" module ^^
I'm interested by your other projects like this one and I have a few questions before buying this sensor.

First, I would like to know if I can swipe pages with it (use with this module for example : https://github.com/edward-shen/MMM-pages). It could be interesting to have modules on another page and to swipe with gesture commands.
Second, how much meters I have to be from the sensor for detection ?

Thanks ! :)

eouia, 2019-06-03:
1) Of course, possible. You can bind your gesture "LEFT" or "RIGHT" to "PAGE_INCREMENT" or "PAGE_DECREMENT" Is this what you need?
2) Not so far. around 10 ~ 20cm.

JohnB7777, 2019-06-03:
Thank you for you quick response !

1. Yes, I think. I would like to change pages (if I have modules on a second page too) using gesture.
2. It's close but it can be enough. Maybe you should try with other sensors allowing us to be further away from the screen (I don't know if it's possible but it might be interesting - maybe with kinect by microsoft ?)

Anyway, I like the feature that permit to stop a video. A few days ago, I run a video with Google Assistant and I couldn't stop it ! It didn't hear me anymore when the video was running

eouia, 2019-06-03:
Frankly said, it is available also with AMK2 itself by `pauseOnYoutube:false` option. But I don't recommend. Because;
- Sound from video itself could make activation Hotword or AMK2 unintentionally. (Real AI speakers can avoid this by using Array Mic to distinguish user's far voice from near speaker output)
- Or just your talking with ppl or even some noise also makes video to stop due to it's sensitivity.

Anyway, I'll remove this YouTube related features on next update of AMK2. I hate this. :(

eouia, 2019-06-03:
Kinnect or other 3D cams or sensor(motion recognition with Deep learning), I'm considering and researching. But RPI has not enough CPU/GPU power to handle them. I'm playing with NVIDIA Jetson Nano and Khadas Edge-V, RockPi m4 and Asus TinkerBoard instead Raspberry Pi.

JohnB7777, 2019-06-03:
You will remove the possibility to start a video on youtube using voice ?
I hope no because I think it's really cool but we have to find a way to make things better.
Maybe there is a way to make google assistant hear us despite the video ?

Yes, indeed, RPI isn't enough powerfull. Maybe in the next version... :)

eouia, 2019-06-03:
I'll separate YouTube related feature to independent module. (It is already released - MMM-YouTube). 
On the next version of AMK2, some more features(Open web url, Showing image, Direction and Map, including Spotify (MMM-Spotify)) will be created as each independent module. AMK2 would be focusing on more Assistant itself and commanding for MM. User's can opt-in the features by configuration to get richer experience by combining modules.

JohnB7777, 2019-06-03:
Ah, ok, I will install this module then.
This is a good idea to separate modules, so we can choose whick install.

JohnB7777, 2019-06-03:
I see your youtube module, can I control it by using MMM-AssistantMk2 ?

eouia, 2019-06-04:
Not yet fully. next AMK2 would have that feature.

closed 2019-06-04 by eouia