MMRIZE/MMM-GroveGestures #13: [GESTURE] Gesture script is finished. [GESTURE] Python script is terminated. It will restart soon. Keyboard Interrupted

StefanMeGit, 2019-06-05:
```
Starting MagicMirror: v2.7.1
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: mvgmunich
Initializing new module helper ...
Module helper loaded: newsfeed
Initializing new module helper ...
Module helper loaded: MMM-NowPlayingOnSpotify
Initializing new module helper ...
Module helper loaded: MMM-GroveGestures
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: mvgmunich
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-NowPlayingOnSpotify
Connecting socket for: MMM-GroveGestures
Sockets connected & modules started ...
Launching application.
Create new news fetcher for url: http://rss.sueddeutsche.de/rss/Topthemen - Interval: 300000
(node:2602) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Refreshed access token because it has expired. Expired at: 21:39:26 now is: 21:39:27
Keyboard Interrupted
[GESTURE] Gesture script is finished.
Refreshed access token because it has expired. Expired at: 22:39:27 now is: 21:39:27
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
[GESTURE] Python script is terminated. It will restart soon.
Keyboard Interrupted
[GESTURE] Gesture script is finished.
```

Directly after start. Did I miss something? The module never shows up. The python test script does only work with SUDO (I can see the gestures in the terminal) `sudo python gesture_print.py`, but if I try `python gesture_print.py` I get the following error: 
```
python gesture_print.py
Traceback (most recent call last):
  File "gesture_print.py", line 37, in <module>
    import grove_gesture_sensor
  File "/home/pi/MagicMirror/modules/MMM-GroveGestures/py/grove_gesture_sensor.py", line 26, in <module>
    bus = smbus.SMBus(1)
IOError: [Errno 13] Permission denied
```

eouia, 2019-06-05:
It means you haven't installed MagicMirror or module with right permission. Usually ppl made a mistake to use `sudo` on `git clone...` or `npm install`. That makes your MM or module should be run with `superuser` permission. 
I don't know which step you've made a mistake. Anyway remove module and try to reinstall carefully without `sudo`.

StefanMeGit, 2019-06-07:
Thanks for you help. Did a clean install of MM. No sudo. MM starts up fine. Just no config created yet.

When I try to test the sensor without sudo, I got the same error. Any idea what went wrong?

```
> magicmirror@2.7.1 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.7.1
Loading config ...
WARNING! Could not find config file. Please create one. Starting with default configuration.
Loading module helpers ...
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: helloworld.
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Sockets connected & modules started ...
Launching application.
^Cpi@RPI3_Mirror:~/MagicMirror $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- 73 -- -- -- --                         
pi@RPI3_Mirror:~/MagicMirror $ cd ~/MagicMirror/modules
pi@RPI3_Mirror:~/MagicMirror/modules $ git clone https://github.com/eouia/MMM-GroveGestures
Klone nach 'MMM-GroveGestures' ...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 69 (delta 1), reused 3 (delta 1), pack-reused 63
Entpacke Objekte: 100% (69/69), Fertig.
pi@RPI3_Mirror:~/MagicMirror/modules $ cd MMM-GroveGestures
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures $ npm install
npm notice created a lockfile as package-lock.json. You should commit this file.
added 1 package in 3.876s
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures $ cd scripts
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/scripts $ chmod +x *.sh
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/scripts $ cd ~/MagicMirror/modules/MMM-GroveGestures/py
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/py $ cp grove_gesture_sensor.py.RPI grove_gesture_sensor.py
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/py $ cd ~/MagicMirror/modules/MMM-GroveGestures/py
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/py $ python gesture_print.py
Traceback (most recent call last):
  File "gesture_print.py", line 37, in <module>
    import grove_gesture_sensor
  File "/home/pi/MagicMirror/modules/MMM-GroveGestures/py/grove_gesture_sensor.py", line 26, in <module>
    bus = smbus.SMBus(1)
IOError: [Errno 13] Permission denied
pi@RPI3_Mirror:~/MagicMirror/modules/MMM-GroveGestures/py $ 

```

eouia, 2019-06-07:
Go to Raspberry config program(raspi-config), Then check which I2C Bus is enabled. I assumed Bus 1 (I2C_1) be enabled. if you have to use other bus, modify python sources to other bus (`SMBUS(n)`)

StefanMeGit, 2019-06-13:
Strange... i made a clean start and now everything works. Thanks for help!

closed 2019-06-13 by StefanMeGit