MMRIZE/MMM-GroveGestures #14: Send more than one notification after gesture

lxne, 2019-10-21:
Hi, is there a way to send two or more notifications after one gesture recognition with GroveGesture?

I have tried
```
        "RIGHT-LEFT": {
                notificationExec: {
                      notification: "NOTIFICATION-FOR-MODULE-A",
                      payload: null,
                },
                notificationExec: {
                      notification: "NOTIFICATION-FOR-MODULE-B",
                      payload: null,
                },
        },
```
Unfortunately only the second notification is being recognized.

Best wishes,
Alex

eouia, 2019-10-22:
Not possible, but there could be some tricky ways.

1) Use `MMM-NotificationTrigger`. It could relay and translate notifications from one to others It could fire multi notifications (or sequence by delay).

2) Or, you can use `moduleExec` to handle multi notifications by the module itself.
like this;
```js
moduleExec: {
  module: ["MMM-GroveGestures"], 
  exec: (module, gestures) => {
    module.sendNotification("NOTIFICATION_A", {somePayloadA: someValueA, ...})
    module.sendNotification("NOTIFICATION_B", {somePayloadB: someValueB, ...})
  }
}
```
or like this;
```js
moduleExec: {
  module: ["MMM-TargetA", "MMM-TargetB"],
  exec: (module, gestures) => {
    if (module.name == "MMM-TargetA") {
      module.notificationReceived("NOTIFICATION_A", {somePayloadA: someValueA, ...})
    }
    if (module.name == "MMM-TargetB") {
      module.notificationReceived("NOTIFICATION_B", {somePayloadB: someValueB, ...})
    }
  }
}
```

Not tested in real machine, but you can catch the idea.

lxne, 2019-10-23:
Thanks for your reply.
I tried the moduleExec way but could not succeed.
```
        "CLOCKWISE": {
          notificationExec: {
            notification: "VOLUME_UP",
            payload: null,
          },
          moduleExec: {
            module: ["MMM-GroveGestures"],
            exec: (module, gestures) => {
              module.sendNotification("SHOW_ALERT", { message: "Volume up.", timer: 5000 })
            }      
          }
        },
```
The VOLUME_UP notification worked as expected but the SHOW_ALERT is not doing anything.
And if I flip the commands SHOW_ALERT is working but VOLUME_UP not.
```
        "CLOCKWISE": {
          notificationExec: {
            notification: "SHOW_ALERT",
            payload:  { message: "Volume up.", timer: 5000 }
          },
          moduleExec: {
            module: ["MMM-GroveGestures"],
            exec: (module, gestures) => {
              module.sendNotification("VOLUME_UP", null)
            }      
          }
        },
```
Any ideas?
NotificationTriggers could be an option but it seems very tedious for the things I want to do.

Best wishes,
Alex

eouia, 2019-10-23:
At this moment, I have no sensor to test. so I cannot test REALLY, Wait a while to prepare the test environment.

eouia, 2019-10-23:
Sorry, there was a bug. I fixed. update and try moduleExec again.

lxne, 2019-10-23:
Thank you eouia!
Now it works.

closed 2019-10-23 by lxne