MMRIZE/MMM-GroveGestures #16: h/w is not installed proberly

8acoustic, 2020-05-27:
not showing matrix 73 pls help

eouia, 2020-05-27:
First check your `raspi-config`. Confirm I2C using is enabled. (Usually I2C_1)
Second confirm your cable is properly connected.

8acoustic, 2020-05-27:
cable are connected properly 
i am getting an isssue while installing 
"sudo apt-get install i2c-tools"

pi@raspberrypi:~ $ sudo apt-get install i2c-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
i2c-tools is already the newest version (4.1-1).
The following package was automatically installed and is no longer required:
  libmicrodns0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
Need to get 0 B/2,641 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
dpkg: error processing package libreoffice-help-common (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
dpkg: dependency problems prevent configuration of libreoffice-help-en-us:
 libreoffice-help-en-us depends on libreoffice-help-common (= 1:6.1.5-3+rpi1+deb10u6+rpt1); however:
  Package libreoffice-help-common is not configured yet.

dpkg: error processing package libreoffice-help-en-us (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libreoffice-help-en-gb:
 libreoffice-help-en-gb depends on libreoffice-help-common (= 1:6.1.5-3+rpi1+deb10u6+rpt1); however:
  Package libreoffice-help-common is not configured yet.

dpkg: error processing package libreoffice-help-en-gb (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libreoffice-help-common
 libreoffice-help-en-us
 libreoffice-help-en-gb
E: Sub-process /usr/bin/dpkg returned an error code (1)

pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

eouia, 2020-05-27:
I have no idea about libreoffice-somethings. I think your RPI was corrupted by the uncompleted installation of some dependencies. (Not related with this module or i2c)
Anyway, you can see matrix 73, now.

8acoustic, 2020-05-28:
yes so no issues will be there right?

eouia, 2020-05-28:
I think so. Doesn't module work?

8acoustic, 2020-06-07:
it works tysm
just figuring out how to change pages with this sensor

eouia, 2020-06-08:
This module itself has no feature of changing pages. This module can only emit customizable notifications(commands). So you need a module that can be controllable by custom notification to change page.

closed 2023-09-04 by eouia