MMRIZE/MMM-GroveGestures #21: Notification for MMM-CalendarExt3

Antislag, 2022-08-07:
Hi, 

first of all thanks for the great module. It works like a charm with my MagicMirror!

I want to add a notification to change the monthly view ("UP": month +1;  "DOWN": month -1).
According to MMM-CalendarExt3 I need to add CX3_GLANCE_CALENDAR, payload: {instanceId, step}.

But if I adapt my config with
                                               "UP": {
                                                notificationExec: {
                                                notification: "CX3_GLANCE_CALENDAR",
                                                payload: {null,1}
                                                                  }
                                                         },

MagicMirror won´t start. 

With following config
                                              "UP": {
                                                notificationExec: {
                                                notification: "CX3_GLANCE_CALENDAR",
                                                payload: 1
                                                                  }
                                                         },

MagicMirror starts but if I swipe UP nothing happens. 
I´m a bloody beginner and don´t know how to proceed. Can you maybe give a hint how to solve the problem?

Thanks in advance!

eouia, 2022-08-08:
I think `payload` was improper. Try this.

```js
payload: { step: 1 }
```

Antislag, 2022-08-08:
Thank you so much! It works perfect.

closed 2022-08-08 by Antislag