MMRIZE/MMM-GroveGestures #22: OnIdle not working anymore

Antislag, 2022-10-18:
Hi,

I just noticed that my screen is not shutting down anymore after 10*60 secs.
Is there a known issue and maybe a solution?
I didnt change anything on the mirror. Just updated the raspy from time to time.

Thanks in advance.

eouia, 2022-10-20:
Other features (Gesture detecting) are working as usual? only OnIdle is not? Could you show me the config of the module?

Antislag, 2022-10-20:
Yes, everything works fine. I can copy my config toda evening.

The function MonitorOff (MMM-Remote-control) is not working aswell. It seems the HDMI deactivation is not working overall.

Antislag, 2022-10-20:
{
                        module: "MMM-GroveGestures",
                        position: "bottom_right",
                        config: {
                                autoStart: true, //When Mirror starts, recognition will start.
                                verbose:false, // If set as `true`, useful messages will be logged.
                                recognitionTimeout: 2000, //Gesture sequence will be ended after this time from last recognized gesture.
                                cancelGesture: "WAVE", //If set, You can cancel gesture sequence with this gesture.
                                visible: true, //Recognized gesture sequence will be displayed on position
                                idleTimer: 10*60*1000, // `0` for disable, After this time from last gesture, onIdle will be executed.
                                onIdle: { // See command section
                                         notificationExec: {
                                                          notification: "REMOTE_ACTION",
                                                            payload: {
                                                                action: "MONITOROFF",
                                                                     },
                                        }
                                },
                                onDetected: {
                                         notificationExec: {
                                                          notification: "REMOTE_ACTION",
                                                            payload: {
                                                                action: "MONITORON",
                                                                     },
                                                },
      /* You can make Mirror to wake up the modules which were hidden by onIdle with any gestures.
      moduleExec: {
        module: [],
        exec: (module) => {
          module.show(1000, null, {lockstring:"GESTURE"})
        }
      }
      */
                                },

                                gestureMapFromTo: { //When your sensor is installed with rotated direction, you can calibrate with this.
                                        "Up": "UP",
                                        "Down": "DOWN",
                                        "Left": "LEFT",
                                        "Right": "RIGHT",
                                        "Forward": "FORWARD",
                                        "Backward": "BACKWARD",
                                        "Clockwise": "CLOCKWISE",
                                        "anti-clockwise": "ANTICLOCKWISE",
                                        "wave": "WAVE"
                                },

                                defaultNotification: "GESTURE",
                                pythonPath: "/usr/bin/python", // your python path

                                defaultCommandSet: "default",
                                commandSet: {
                                        "default": {
                                                "FORWARD-BACKWARD": {
                                                notificationExec: {
                                                notification: "ASSISTANT_ACTIVATE",
                                                payload: null
                                                                  }
                                                                    },
                                                "LEFT-RIGHT": {
                                                notificationExec: {
                                                notification: "ASSISTANT_CLEAR",
                                                payload:null,
                                                                  }
                                                              },
                                                "CLOCKWISE": {
                                                moduleExec: {
                                                module: [],
                                                exec: (module, gestures) => {
                                                module.hide(1000, null, {lockstring:"GESTURE"})
[... 56 lines cut ...]
                                        /*      "DOWN-DOWN-DOWN": {
                                                        notificationExec: {
                                                          notification: "REMOTE_ACTION",
                                                            payload: {
                                                                action: "SHUTDOWN",
                                                                     },
                                                                },
                                                        },
                                                "UP-UP-UP": {
                                                        notificationExec: {
                                                          notification: "REMOTE_ACTION",
                                                            payload: {
                                                                action: "REBOOT",
                                                                     },  // you don't need to refer to a module name for the MONITORONN action (only necessa>
[... cut ...]
                                                       },*/

                                        },
                                        commandSetTrigger: {
                                                "DEFAULT_MODE": "DEFAULT_MODE",
                                                "PAGE_NUMBER_IS": (payload) => { return payload } // this is what set the right command set based on MMM-Pages feedback "PAGE_NUMBER_IS"
                                        },
    },
  }
                },

eouia, 2022-10-21:
I think it was not this module's fault.
You can test onIdle feature itself like this;
```js
onIdle: {
  notificationExec: {
    notification: "SHOW_ALERT",
    payload: {
      title: "onIdle Test",
      message: "This message should be displayed when onIdle feature is working",
      timer: 10*1000,
      type: "notification"
    },
  }
},
```

Antislag, 2022-10-21:
Ok, I will try the onIdle today. Thanks alot for your answers!

Antislag, 2022-10-22:
Yes, onIdle works. Need to find another somution for MonitorOff. Thanks for your Support!

closed 2022-10-22 by Antislag

Antislag, 2022-10-22:
Sorry to bother you again but I can´t find a way to solve this problem.
Is there a build-in command to shutdown the monitor, which I can use? 
I´ve seen there are 2 scripts in the folder. How can i use these?