MMRIZE/MMM-GroveGestures #23: monitor on_off

alexsquared, 2023-05-07:
in looking at your scripts folder, I see scripts to turn the monitor on/off via the HDMI port.   How do I execute those scripts?   Or is there another command that I can use to turn the monitor output on/off? Otherwise, I just got it working and really enjoy it.    do you have any other command sets that are useful?  Ie hiding/showing certain modules instead of all of them?

eouia, 2023-05-07:
You can assign `command` to specific `gesture`.

`gesture` is your input. it could be one or sequential series of gestures.
`command` is what you want this module to do with that gesture. There could be 3 types of commands
- executing shell script. (like turn off screen. you can execute any shell script/command if no harm.)
- emitting notification (You can deliver some notification to other module, like `SHOW_ALERT` notification to the default `alert` module)
- Or, handling other modules directly. (as far as you know the structure of that target module, / For expert.)

`scripts` folder is just one example. I prepared there a simple shell script to execute `vcgencmd` of legacy RPI environment. Maybe under other environments, it may not work. 

Anyway, to execute it with the gesture 'LEFT-RIGHT' for example;
just describe it in your config like this. (I'll assume you will use only `default` `commandSet`)

```js
// ...
commandSet: {
  "default" : {
    "LEFT-RIGHT" : {
      shellExec: "./scripts/screenoff.sh",
    },
    // ... other gestures definition ...
  },
},
```
When LEFT-RIGHT gesture is detected, this module will try to execute shell script "./scripts/screenoff.sh".

If you want to hiding/showing specific modules not all, you can also use `moduleExec`.

```js
// ... other gesture definitions, ...
"UP" : {
  moduleExec : {
    module: ["clock", "weather"],
    exec: (module) => {
      module.hide(1000, null, {lockstring: "BY_GROVE_GESTURE"})
    }
  },
},
```
This makes, when your gesture is "UP", the modules "clock" and "weather" will be hidden with lockstring. I think you can understand how it goes.

alexsquared, 2023-06-09:
This does help.  thank you very much.

closed 2023-06-09 by eouia