MMRIZE/MMM-GroveGestures #24: controls not initiated

alexsquared, 2023-07-14:
I have the hardware and module installed.  Took a little bit of understanding to get it up and running but ran it for months with no issue.   With no other changes to the system, it has randomly stopped executing the commands to the modules.   I only have four gestures programmed.  up/down/left/right.  One hides all my modules, one shows all my modules, one hides a specific module and one shows that specific module.   all was working well.   suddenly my commands no longer worked.   When I initiate the gesture, I see the gesture recognized through the notification on the screen, however, the task isn't executed and no state change to the other modules occurs.   is there a way to troubleshoot this?

eouia, 2023-07-14:
What changed? Something updated? OS, MagicMirror, module or anything? Something new installed?
- Describe what you have: HW, OS version, node & npm version, MagicMirror version.
- Can you provide backend log and frontend log to me?
- Show me your current config
- Try the simplest setting (Remove all other 3rd party modules except GroveGesture and default modules)

alexsquared, 2023-07-14:
I will grab my config and share on here.    HW = Raspberry Pi 3B+, OS is Bullseye, MagicMirror version is 2.24, node/npm version not sure, but will grab. 

nothing has changed in my config/OS/etc, which makes it very hard to track.   I have noticed that I'm regularly losing internet connection on the Pi now too, but I don't know if that is related to the module or not.

alexsquared, 2023-07-14:
`module: "MMM-GroveGestures",
  position: "bottom_left",
  config: {
    autoStart: true, //When Mirror starts, recognition will start.
    verbose:true, // If set as `true`, useful messages will be logged.
    recognitionTimeout: 1000, //Gesture sequence will be ended after this time from last recognized gesture.
    cancelGesture: "WAVE", //If set, You can cancel gesture sequence with this gesture.
    visible: true, //Recognized gesture sequence will be displayed on position

    idleTimer: 1000*60*30, // `0` for disable, After this time from last gesture, onIdle will be executed.
    onIdle: { // See command section
      moduleExec: {
        module: [],
        exec: (module, gestures) => {
          module.hide(1000, null, {lockstring:"GESTURE"})
        }
      }
    },
    onDetected: {
      notificationExec: {
        notification: "GESTURE_DETECTED",
      },
      /* You can make Mirror to wake up the modules which were hidden by onIdle with any gestures.
      moduleExec: {
        module: [],
        exec: (module) => {
          module.show(1000, null, {lockstring:"GESTURE"})
        }
      }
      */
    },

    gestureMapFromTo: { //When your sensor is installed with rotated direction, you can calibrate with this.
      "Up": "UP",
      "Down": "DOWN",
      "Left": "LEFT",
      "Right": "RIGHT",
      "Forward": "FORWARD",
      "Backward": "BACKWARD",
      "Clockwise": "CLOCKWISE",
      "anti-clockwise": "ANTICLOCKWISE",
      "wave": "WAVE"
    },

    defaultNotification: "GESTURE",
    pythonPath: "/usr/bin/python", // your python path

    defaultCommandSet: "default",
    commandSet: {
      "default": 
     //   {
     //   "FORWARD-BACKWARD": {
     //     notificationExec: {
     //       notification: "ASSISTANT_ACTIVATE",
     //       payload: null
      //    }
     //   },
     //   "LEFT-RIGHT": {
     //     notificationExec: {
     //       notification: "ASSISTANT_CLEAR",
     //       payload:null,
      //    }
      //  },
        {
        "UP": {
          moduleExec: {
            module: [],
            exec: (module, gestures) => {
              module.hide(1000, null, {lockstring:"GESTURE"})
            }
          }
        },
        "DOWN": {
          moduleExec: {
            module: [],
            exec: (module, gestures) => {
              module.show(1000, null, {lockstring:"GESTURE"})
            }
          }
        },
        "LEFT": {
          moduleExec: {
            module: ["MMM-WebView"],
            exec: (module, gestures) => {
              module.show(1000, null, {lockstring:"GESTURE"})
          }
	}
        },
        "RIGHT": {
	  moduleExec: {
            module: ["MMM-WebView"],
            exec: (module, gestures) => {
              module.hide(1000, null, {lockstring:"GESTURE"})
          }
	}
        },`

alexsquared, 2023-07-14:
not the entire config, but the grove gesture section specifically.

alexsquared, 2023-07-14:
output from NPM list. 

├── colors@1.4.0
├── console-stamp@3.1.1
├── digest-fetch@2.0.3
├── electron@25.2.0
├── envsub@4.1.0
├── UNMET DEPENDENCY eslint-config-prettier@^8.8.0
├── UNMET DEPENDENCY eslint-plugin-import@^2.27.5
├── UNMET DEPENDENCY eslint-plugin-jest@^27.2.2
├── UNMET DEPENDENCY eslint-plugin-jsdoc@^46.4.2
├── UNMET DEPENDENCY eslint-plugin-prettier@^4.2.1
├── eslint@8.43.0
├── UNMET DEPENDENCY express-basic-auth@^1.2.1
├── express-ipfilter@1.3.1
├── express@4.18.2
├── feedme@2.0.2
├── helmet@7.0.0
├── UNMET DEPENDENCY husky@^8.0.3
├── iconv-lite@0.6.3
├── UNMET DEPENDENCY jest@^29.5.0
├── UNMET DEPENDENCY jsdom@^22.1.0
├── lodash@4.17.21
├── luxon@1.28.1
├── module-alias@2.2.3
├── moment@2.29.4
├── node-fetch@2.6.12
├── node-ical@0.16.1
├── UNMET DEPENDENCY playwright@^1.35.1
├── UNMET DEPENDENCY prettier@^2.8.8
├── UNMET DEPENDENCY pretty-quick@^3.1.3
├── UNMET DEPENDENCY sinon@^15.2.0
├── socket.io@4.7.1
├── UNMET DEPENDENCY stylelint-config-standard@^33.0.0
├── UNMET DEPENDENCY stylelint-prettier@^3.0.0
├── UNMET DEPENDENCY stylelint@^15.9.0
└── UNMET DEPENDENCY suncalc@^1.9.0

npm ERR! code ELSPROBLEMS
npm ERR! missing: eslint-config-prettier@^8.8.0, required by magicmirror@2.24.0
npm ERR! missing: eslint-plugin-import@^2.27.5, required by magicmirror@2.24.0
npm ERR! missing: eslint-plugin-jest@^27.2.2, required by magicmirror@2.24.0
npm ERR! missing: eslint-plugin-jsdoc@^46.4.2, required by magicmirror@2.24.0
npm ERR! missing: eslint-plugin-prettier@^4.2.1, required by magicmirror@2.24.0
npm ERR! missing: express-basic-auth@^1.2.1, required by magicmirror@2.24.0
npm ERR! missing: husky@^8.0.3, required by magicmirror@2.24.0
npm ERR! missing: jest@^29.5.0, required by magicmirror@2.24.0
npm ERR! missing: jsdom@^22.1.0, required by magicmirror@2.24.0
npm ERR! missing: playwright@^1.35.1, required by magicmirror@2.24.0
npm ERR! missing: prettier@^2.8.8, required by magicmirror@2.24.0
npm ERR! missing: pretty-quick@^3.1.3, required by magicmirror@2.24.0
npm ERR! missing: sinon@^15.2.0, required by magicmirror@2.24.0
npm ERR! missing: stylelint-config-standard@^33.0.0, required by magicmirror@2.2                                                                 4.0
npm ERR! missing: stylelint-prettier@^3.0.0, required by magicmirror@2.24.0
npm ERR! missing: stylelint@^15.9.0, required by magicmirror@2.24.0
npm ERR! missing: suncalc@^1.9.0, required by magicmirror@2.24.0

npm ERR! A complete log of this run can be found in:

alexsquared, 2023-07-14:
the NPM version is 8.19.4.

eouia, 2023-07-14:
Stop “pm2”(if you are running MmM with it), then try “npm run start:dev” . The terminal will show you what’s going on backend and your MM screen also will show front dev-console. See what both console spit out any suspicious messages.

alexsquared, 2023-07-14:
magicmirror@2.24.0 start:dev
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js dev

[14.07.2023 13:10.25.613] [LOG]   Starting MagicMirror: v2.24.0
[14.07.2023 13:10.25.629] [LOG]   Loading config ...
[14.07.2023 13:10.25.637] [DEBUG] config template file not exists, no envsubst
[14.07.2023 13:10.25.651] [LOG]   Loading module helpers ...
[14.07.2023 13:10.25.654] [LOG]   No helper found for module: alert.
[14.07.2023 13:10.25.698] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.25.700] [LOG]   Module helper loaded: updatenotification
[14.07.2023 13:10.25.702] [LOG]   No helper found for module: clock.
[14.07.2023 13:10.26.229] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.26.231] [LOG]   Module helper loaded: calendar
[14.07.2023 13:10.26.234] [LOG]   No helper found for module: compliments.
[14.07.2023 13:10.27.109] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.27.111] [LOG]   Module helper loaded: MMM-DailyBibleVerse
[14.07.2023 13:10.27.112] [LOG]   No helper found for module: MMM-WeatherGraph.
[14.07.2023 13:10.27.447] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.27.449] [LOG]   Module helper loaded: MMM-Remote-Control
[14.07.2023 13:10.28.004] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.28.005] [LOG]   Module helper loaded: MMM-GooglePhotos
[14.07.2023 13:10.28.007] [LOG]   No helper found for module: MMM-WebView.
[14.07.2023 13:10.28.023] [LOG]   Initializing new module helper ...
[14.07.2023 13:10.28.024] [LOG]   Module helper loaded: MMM-GroveGestures
[14.07.2023 13:10.28.025] [LOG]   All module helpers loaded.
[14.07.2023 13:10.28.048] [LOG]   Starting server on port 8080 ...
[14.07.2023 13:10.28.083] [LOG]   Server started ...
[14.07.2023 13:10.28.086] [LOG]   Connecting socket for: updatenotification
[14.07.2023 13:10.28.087] [LOG]   Starting module helper: updatenotification
[14.07.2023 13:10.28.089] [LOG]   Connecting socket for: calendar
[14.07.2023 13:10.28.091] [LOG]   Starting node helper for: calendar
[14.07.2023 13:10.28.092] [LOG]   Connecting socket for: MMM-DailyBibleVerse
[14.07.2023 13:10.28.093] [LOG]   Started node_helper.js for MMM-DailyBibleVerse.
[14.07.2023 13:10.28.095] [LOG]   Connecting socket for: MMM-Remote-Control
[14.07.2023 13:10.28.097] [LOG]   Starting node helper for: MMM-Remote-Control
[14.07.2023 13:10.28.110] [LOG]   Connecting socket for: MMM-GooglePhotos
[14.07.2023 13:10.28.112] [LOG]   Connecting socket for: MMM-GroveGestures
[14.07.2023 13:10.28.114] [LOG]   Sockets connected & modules started ...
[14.07.2023 13:10.28.750] [LOG]   Launching application.
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
../../buildtools/third_party/libc++/trunk/include/vector:1471: assertion __n < size() failed: vector[] index out of bounds[11535:0714/131032.949767:ERROR:gpu_process_host.cc(                                                               954)] GPU process exited unexpectedly: exit_code=6
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
../../buildtools/third_party/libc++/trunk/include/vector:1471: assertion __n < size() failed: vector[] index out of bounds[11535:0714/131034.330218:ERROR:command_buffer_proxy                                                               _impl.cc(325)] GPU state invalid after WaitForGetOffsetInRange.
[11535:0714/131034.337204:ERROR:gpu_process_host.cc(954)] GPU process exited unexpectedly: exit_code=6
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
../../buildtools/third_party/libc++/trunk/include/vector:1471: assertion __n < size() failed: vector[] index out of bounds[11586:0714/131036.090306:ERROR:command_buffer_proxy                                                               _impl.cc(325)] GPU state invalid after WaitForGetOffsetInRange.
[11586:0714/131036.093787:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[11535:0714/131036.123260:ERROR:gpu_process_host.cc(954)] GPU process exited unexpectedly: exit_code=6
[11579:0714/131037.074609:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[11586:0714/131037.103273:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[14.07.2023 13:10.39.747] [LOG]   Create new calendarfetcher for url: http://www.calendarlabs.com/templates/ical/US-Holidays.ics - Interval: 3600000
[14.07.2023 13:10.39.807] [LOG]   Create new calendarfetcher for url: https://calendar.google.com/calendar/ical/sk8vqq6a879dev8qer0j1r5h64%40group.calendar.google.com/private                                                               -f6dabff35c95397ed9434ed90ad83f68/basic.ics - Interval: 3600000
[14.07.2023 13:10.40.038] [LOG]   MMM-DailyBibleVerse node helper received a socket notification: START - Payload: NASB
[14.07.2023 13:10.40.101] [INFO]  Checking git for module: MMM-DailyBibleVerse
[14.07.2023 13:10.40.479] [INFO]  Checking git for module: MMM-WeatherGraph
[14.07.2023 13:10.41.224] [LOG]   [GESTURE] ORIGIN:wake-up finish.
[... cut ...]
[14.07.2023 13:11.55.322] [LOG]   [GESTURE] Gesture script is finished.
[14.07.2023 13:11.56.192] [LOG]   [GESTURE] Python script is terminated. It will restart soon.
[14.07.2023 13:11.57.003] [LOG]   Keyboard Interrupted
[14.07.2023 13:11.57.006] [LOG]   [GESTURE] Gesture script is finished.
[14.07.2023 13:11.57.509] [LOG]   [GESTURE] Python script is terminated. It will restart soon.
[14.07.2023 13:11.58.303] [LOG]   [GESTURE] ORIGIN:wake-up finish.
[14.07.2023 13:11.58.398] [LOG]   [GESTURE] ORIGIN:Paj7620 initialize register finished.

alexsquared, 2023-07-14:
I then went and initiated some gestures. 

[14.07.2023 13:11.58.398] [LOG]   [GESTURE] ORIGIN:Paj7620 initialize register finished.
[14.07.2023 13:15.22.411] [LOG]   [GESTURE] ORIGIN:Left
[14.07.2023 13:15.22.414] [LOG]   [GESTURE] Ongoing: LEFT
[14.07.2023 13:15.22.417] [LOG]   [GESTURE] Timer reset
[14.07.2023 13:15.23.421] [LOG]   [GESTURE] Finish: LEFT
[14.07.2023 13:15.26.823] [LOG]   [GESTURE] ORIGIN:Right
[14.07.2023 13:15.26.825] [LOG]   [GESTURE] Ongoing: RIGHT
[14.07.2023 13:15.26.829] [LOG]   [GESTURE] Timer reset
[14.07.2023 13:15.27.833] [LOG]   [GESTURE] Finish: RIGHT
[14.07.2023 13:15.30.959] [LOG]   [GESTURE] ORIGIN:Right
[14.07.2023 13:15.30.961] [LOG]   [GESTURE] Ongoing: RIGHT
[14.07.2023 13:15.30.963] [LOG]   [GESTURE] Timer reset
[14.07.2023 13:15.31.967] [LOG]   [GESTURE] Finish: RIGHT

eouia, 2023-07-14:
Seems module itself working, nothing wrong. Hmmmm.
How about front dev-console? Can you check it also?

alexsquared, 2023-07-14:
looked and I don't see any errors or anything in dev-console on the display.    I went and did gestures and no outputs occurred on dev-console.

eouia, 2023-07-15:
ATM I have no sensor to test, but I’ll look inside more.

alexsquared, 2023-07-21:
apologies for the delay as I didn't have a chance to get back on this for several days.   are you wanting me to run this and grab log output data?

eouia, 2023-07-22:
Yes. It could be a help for diagnosing.