MMRIZE/MMM-GroveGestures #25: 5-pin sensor to Raspberry Pi?

rmcon, 2023-09-10:
The sensor I purchased has 5 pins, not 4 as in your diagram.  The 5th pin is labeled on the sensor as INT and the description online says "Interrupt output, connects to I/O pin".  The other 4 pins are connected per your specification (pin 1 "VCC" to RPI pin 4, pin 2 "GND" to RPI pin 6, pin 3 "SDA" to RPI pin 3, pin 4 "SCL" to RPI pin 5) and I've left this 5th wire disconnected.  The brand is CQRobot 3D Gesture Sensor, purchased from the major online retailer here in the US.

I haven't had any luck getting the sensor to respond yet and I'm not sure if I just don't have it configured correctly or if the 5th pin being disconnected could be preventing the sensor from working?  The documentation on the manufacturer's website isn't clear whether it has to be connected, as the diagrams show the wire not going to any particular pin, not sure if that means that it's optional to connect or just not depicted because there are multiple options.  Thoughts?

eouia, 2023-09-12:
I think the chipset itself might be compatible, but the connection is different so the main script should be rewritten. Unfortunately, I don’t have experience with that sensor, so cannot help.

rmcon, 2023-09-16:
Sorry for the delayed response, I was on travel.  After my original post I discovered the test script (print.py) for the sensor and it worked beautifully with this sensor.  I discovered that the orientation of the sensor was 90 degrees off, so I adjusted that in the Config file, but still no luck.  That makes me hopeful that this sensor will still work and I just need to figure out what's up with the settings.  I'll dig deeper into the previous issues and try to find a solution.

eouia, 2023-09-18:
I think ` gestureMapFromTo` would be for that purpose.

closed 2024-04-29 by eouia

cbass27, 2025-08-05:
@rmcon I know this is a late reply to an already closed issue, but it sounds like you were able to get your CQRobot gesture sensor working as I have the same one? Did you have to modify any of the python scripts to get yours working? I did have to use different pins on the Raspberry Pi 5 I have because of the documented suggested pins I have in use already. The device shows as available when I run i2cdetect. Thanks!