MMRIZE/MMM-GroveGestures #27: Controls no longer working

alexsquared, 2024-11-08:
This is a continuation of the previously closed issue linked below. 
I had to do a fresh install as my last MM was hanging up.   I installed the module, enabled I2c, and controls still don't work.   I get the gesture response (ie left, down, etc) to show up on my screen as I have that enabled, but the subsequent action (hide modules) doesn't work.   Please note that previously this worked and then suddenly stopped.   a lot of troubleshooting was done, however, we were never able to reestablish control before I had to do the fresh install.    Any help would be greatly appreciated.   The functionality provided by this app has been great when it works.   

https://github.com/MMRIZE/MMM-GroveGestures/issues/24

eouia, 2024-11-11:
With dev-debug mode, you can get some suspicious warning or error messages from MM on the front-side. 
```
npm run start:dev
```

Or you can enter dev-console on running with <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>J</kbd>

Could you see any suspicious message there when you try to manipulate?

alexsquared, 2024-11-11:
does it matter whether I'm using pm2 or npm?

lxne, 2024-11-12:
Did you test the sensor?
```
cd ~/MagicMirror/modules/MMM-GroveGestures/py
python gesture_print.py
```

alexsquared, 2024-11-14:
yup, it's running fine. 

raspberrypi:~ $ cd ~/MagicMirror/modules/MMM-GroveGestures/py
python gesture_print.py
wake-up finish.
Paj7620 initialize register finished.
Down
Left

alexsquared, 2024-11-14:
> Both doesn't matter, but if you are using pm2 currently, you have to stop your previous instance of MM on PM2. Then you can run `npm run start:dev`. Or just open dev-console on running by shortcut key.

When i stop MM via PM2 and then run NPM run start:dev, I get the below. 

npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/a2/package.json
npm ERR! errno -2
npm ERR! enoent Could not read package.json: Error: ENOENT: no such file or directory, open '/home/a2/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in: /home/a2/.npm/_logs/2024-11-14T21_57_02_006Z-debug-0.log

alexsquared, 2024-11-14:
`0 verbose cli /usr/local/bin/node /usr/local/bin/npm
1 info using npm@10.1.0
2 info using node@v20.9.0
3 timing npm:load:whichnode Completed in 5ms
4 timing config:load:defaults Completed in 9ms
5 timing config:load:file:/usr/local/lib/node_modules/npm/npmrc Completed in 2ms
6 timing config:load:builtin Completed in 3ms
7 timing config:load:cli Completed in 13ms
8 timing config:load:env Completed in 2ms
9 timing config:load:project Completed in 6ms
10 timing config:load:file:/home/a2/.npmrc Completed in 1ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:/usr/local/etc/npmrc Completed in 22ms
13 timing config:load:global Completed in 22ms
14 timing config:load:setEnvs Completed in 5ms
15 timing config:load Completed in 66ms
16 timing npm:load:configload Completed in 68ms
17 timing config:load:flatten Completed in 14ms
18 timing npm:load:mkdirpcache Completed in 1ms
19 timing npm:load:mkdirplogs Completed in 1ms
20 verbose title npm run start:dev
21 verbose argv "run" "start:dev"
22 timing npm:load:setTitle Completed in 4ms
23 timing npm:load:display Completed in 3ms
24 verbose logfile logs-max:10 dir:/home/a2/.npm/_logs/2024-11-14T21_57_02_006Z-
25 verbose logfile /home/a2/.npm/_logs/2024-11-14T21_57_02_006Z-debug-0.log
26 timing npm:load:logFile Completed in 49ms
27 timing npm:load:timers Completed in 0ms
28 timing npm:load:configScope Completed in 0ms
29 timing npm:load Completed in 236ms
30 timing command:run Completed in 14ms
31 verbose stack Error: Could not read package.json: Error: ENOENT: no such file or directory, open '/home/a2/package.json'
32 verbose cwd /home/a2
33 verbose Linux 6.6.51+rpt-rpi-v7
34 verbose node v20.9.0
35 verbose npm  v10.1.0
36 error code ENOENT
37 error syscall open
38 error path /home/a2/package.json
39 error errno -2
40 error enoent Could not read package.json: Error: ENOENT: no such file or directory, open '/home/a2/package.json'
41 error enoent This is related to npm not being able to find a file.
41 error enoent
42 verbose exit -2
43 timing npm Completed in 718ms
44 verbose code -2
45 error A complete log of this run can be found in: /home/a2/.npm/_logs/2024-11-14T21_57_02_006Z-debug-0.log
`

eouia, 2024-11-15:
You have to run `npm` command in your MagicMirror directory. Sorry, my explanation was not enough.

eouia, 2024-11-16:
It seems your sensor is working.
Let's check the command thing. Could you show me the config of the module?

alexsquared, 2024-11-18:
{
  module: "MMM-GroveGestures",
  position: "bottom_left",
  config: {
    autoStart: true, //When Mirror starts, recognition will start.
    verbose:true, // If set as `true`, useful messages will be logged.
    recognitionTimeout: 1000, //Gesture sequence will be ended after this time from last recognized gesture.
    cancelGesture: "WAVE", //If set, You can cancel gesture sequence with this gesture.
    visible: true, //Recognized gesture sequence will be displayed on position

    idleTimer: 1000*60*30, // `0` for disable, After this time from last gesture, onIdle will be executed.
    onIdle: { // See command section
      moduleExec: {
        module: [],
        exec: (module, gestures) => {
          module.hide(1000, null, {lockstring:"GESTURE"})
        }
      }
    },
    onDetected: {
      notificationExec: {
        notification: "GESTURE_DETECTED",
      },
      /* You can make Mirror to wake up the modules which were hidden by onIdle with any gestures.
      moduleExec: {
        module: [],
        exec: (module) => {
          module.show(1000, null, {lockstring:"GESTURE"})
        }
      }
      */
    },

    gestureMapFromTo: { //When your sensor is installed with rotated direction, you can calibrate with this.
      "Up": "UP",
      "Down": "DOWN",
      "Left": "LEFT",
      "Right": "RIGHT",
      "Forward": "FORWARD",
      "Backward": "BACKWARD",
      "Clockwise": "CLOCKWISE",
      "anti-clockwise": "ANTICLOCKWISE",
      "wave": "WAVE"
    },

    defaultNotification: "GESTURE",
    pythonPath: "/usr/bin/python", // your python path

    defaultCommandSet: "default",
    commandSet: {
      "default":
     //   {
     //   "FORWARD-BACKWARD": {
 //     notificationExec: {
     //       notification: "ASSISTANT_ACTIVATE",
     //       payload: null
      //    }
     //   },
     //   "LEFT-RIGHT": {
     //     notificationExec: {
     //       notification: "ASSISTANT_CLEAR",
     //       payload:null,
      //    }
      //  },
        {
        "UP": {
          moduleExec: {
            module: [],
            exec: (module, gestures) => {
              module.hide(1000, null, {lockstring:"GESTURE"})
            }
          }
        },
        "DOWN": {
          moduleExec: {
            module: [],
            exec: (module, gestures) => {
              module.show(1000, null, {lockstring:"GESTURE"})
            }
          }
        },
        "LEFT": {
          moduleExec: {
            module: ["MMM-WebView"],
            exec: (module, gestures) => {
              module.show(1000, null, {lockstring:"GESTURE"})
          }
        }
        },
        "RIGHT": {
          moduleExec: {
            module: ["MMM-WebView"],
            exec: (module, gestures) => {
              module.hide(1000, null, {lockstring:"GESTURE"})
          }
        }

alexsquared, 2024-11-20:
Any thoughts?

eouia, 2024-11-22:
As I said before; I need the front-end dev-console log, but you showed me the backend-log;
the front-end dev-console should be something like this;
(The right side panel of the MM screen)
[image]

There could be some suspicious error or warning message; 

And to make things simple;
- Backup your current config; then make or copy a new one to test.
- Remove all other modules except `clock` and `MMM-GroveGestures`
- Then, make a simple test configuration to check whether the module works. (e.g. Hide/Show the clock module only)
- Works?
  - If so, your other module makes something wrong.
  - If not working, Let's start to discuss further

alexsquared, 2024-12-04:
apologies for the delay.  we've been busy with work and the holidays, but hope to test in the coming weeks.

alexsquared, 2024-12-04:
had a spare minute.   I opened the dev console and did a gesture 2x.  

[image]

alexsquared, 2024-12-04:
both times I received this error. 

[image]

eouia, 2024-12-05:
I found what was the problem.

**QUICK SOLUTION**
```js
/* module.hide(1000, null, {lockString: "GESTURE" })  <-- NOT VALID ANYMORE */
module.hide(1000, () => { }, { lockString: "GESTURE" })
```

**WHY?**
In the MagicMirror development documentation, it is mentioned that the second parameter, `callback`, in the `module.hide/show` method is optional. This was indeed the case in the past. However, at some point in a later version, the `callback` parameter became mandatory (for reasons unknown, and personally, I think it was an unnecessary change).  
As a result, you can no longer pass `null` as the second parameter. Instead, you need to provide an empty function (`() => {}`).

alexsquared, 2024-12-06:
sounds good.  I will try that and see if that works.   has this been edited in the sample config for the module as well?

alexsquared, 2024-12-06:
well.  Success!  changing that in the config worked like a charm.   thank you so much!   I do have one last question.   Have you come up with a decorative means of covering these, but it still allows the gesture pickup?   I have a wooden frame around our mirror and had to drill and notch a hole for the sensor.   At first I had a small round hole drilled (about 1/2") and had the sensor behind the wood frame with just the sensor pickup exposed, however, it wouldn't pick up gestures.   I'm not sure if it was the shadow cast by the frame or what, but the moment I notched the hole and brought the sensor to the face, it would pick up gestures.   The only downside is now it looks pretty ugly.  I'd like to clean up the look and I'm wondering if I could cover the sensor and the opening to make it decorative but still have it work.

alexsquared, 2024-12-06:
Here is an image with the sensor at the top of the mirror frame. 

[image]

eouia, 2024-12-10:
This sensor operates based on the principles of electromagnetic field theory. If there are no elements disrupting the flow of the electromagnetic field, it should work as expected. However, it may be affected if there are nearby metal frames or other objects that could interfere with the electromagnetic field's flow.

Additionally, in the case of spy mirrors, the metallic components used in the mirror's coating may prevent the sensor from functioning properly behind such surfaces. 

On the other hand, materials like wood, paper, and plastic are unlikely to have a significant impact on its performance. However, since the sensor's range is not very extensive, it may fail to recognize gestures made from a considerable distance.

alexsquared, 2024-12-12:
got it.  I will experiment with some materials on hand and see if I can find something to use to dress it up.

alexsquared, 2024-12-18:
as an FYI, This item can be closed.

commit ecf7595b (2025-07-22): fix: update module.hide to use a callback

This will fix #27. It's necessary to be  compatible with newer MagicMirror versions.
closed 2025-07-22 by KristjanESPERANTO